Koenkk/Z-Stack-firmware #494: firmware.patch for 7.10.02.23

jasondefuria, 2024-02-24:
Hi-

I am trying to build for the newest firmware, and I can't get the latest firmware.patch (from the 7.10.00.98 20230507 release), I'm assuming it needs to be refactored for 7.10.02.23.

I get the following when trying to run:
```
git apply firmware.patch --ignore-space-change
firmware.patch:331: trailing whitespace.
  
firmware.patch:333: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since 
firmware.patch:1397: trailing whitespace.
    // Set TXRX pin to 0 in RX and high impedance in idle/TX. 
firmware.patch:1415: trailing whitespace.
    // Set TXRX pin to 0 in RX/TX and high impedance in idle. 
firmware.patch:1616: new blank line at EOF.
+
error: patch failed: workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg:4
error: workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg: patch does not apply
```

Removing the whitespace does not seem to fix the issue. Any thoughts on how to modify to build a firmware based upon the latest release?

amidarius, 2024-03-20:
Adding --reject to the command line resulted into this output :

```
$ git apply firmware.patch --ignore-space-change --reject
firmware.patch:331: trailing whitespace.

firmware.patch:333: trailing whitespace.
  // ZDSecMgrPermitJoining is never called. Joining via a router would always fail now since
firmware.patch:1397: trailing whitespace.
    // Set TXRX pin to 0 in RX and high impedance in idle/TX.
firmware.patch:1415: trailing whitespace.
    // Set TXRX pin to 0 in RX/TX and high impedance in idle.
Checking patch source/ti/zstack/boards/cc13x2_cc26x2/cc13x2_cc26x2_tirtos7_ticlang.cmd...
warning: source/ti/zstack/boards/cc13x2_cc26x2/cc13x2_cc26x2_tirtos7_ticlang.cmd has type 100644, expected 100755
Checking patch source/ti/zstack/boards/cc13x2x7_cc26x2x7/cc13x2x7_cc26x2x7_tirtos7_ticlang.cmd...
warning: source/ti/zstack/boards/cc13x2x7_cc26x2x7/cc13x2x7_cc26x2x7_tirtos7_ticlang.cmd has type 100644, expected 100755
Checking patch source/ti/zstack/mt/mt.h...
Checking patch source/ti/zstack/mt/mt_util.c...
Checking patch source/ti/zstack/mt/mt_version.c...
Checking patch source/ti/zstack/mt/mt_zdo.c...
Checking patch source/ti/zstack/stack/Config/preinclude.h...
Checking patch source/ti/zstack/stack/af/af.c...
Checking patch source/ti/zstack/stack/nwk/nwk_globals.c...
Checking patch source/ti/zstack/stack/nwk/nwk_util.c...
Checking patch source/ti/zstack/stack/nwk/nwk_util.h...
Checking patch source/ti/zstack/stack/sys/zglobals.c...
Checking patch source/ti/zstack/stack/zdo/zd_app.c...
Checking patch source/ti/zstack/stack/zdo/zd_object.c...
Checking patch source/ti/zstack/stack/zdo/zd_sec_mgr.c...
Checking patch source/ti/zstack/stack/zdo/zd_sec_mgr.h...
Checking patch source/ti/zstack/startup/zstackstartup.c...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/.project...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/Stack/Config/znp_cnf.opts...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/ti_devices_config.c...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/ti_drivers_config.h...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/ti_radio_config.c...
firmware.patch:1616: new blank line at EOF.
+
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/ti_radio_config.h...
Checking patch workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg...
error: while searching for:
 * @cliArgs --board "/ti/boards/CC1352P_2_LAUNCHXL" --rtos "tirtos7" --product "simplelink_cc13xx_cc26xx_sdk@7.10.00.98"
 * @versions {"tool":"1.15.0+2826"}
 */

/**
 * Import the modules used in this configuration.

error: patch failed: workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg:4
Hunk #2 succeeded at 65 (offset -4 lines).
Hunk #3 succeeded at 88 (offset -4 lines).
Hunk #4 succeeded at 179 (offset -5 lines).
Checking patch workspace/znp_CC26X2R1_LAUNCHXL_tirtos7_ticlang/.project...
Checking patch workspace/znp_CC26X2R1_LAUNCHXL_tirtos7_ticlang/Stack/Config/znp_cnf.opts...
Checking patch workspace/znp_CC26X2R1_LAUNCHXL_tirtos7_ticlang/znp.syscfg...
Hunk #3 succeeded at 172 (offset -1 lines).
Checking patch workspace/znp_LP_CC1352P7_4_tirtos7_ticlang/.project...
Checking patch workspace/znp_LP_CC1352P7_4_tirtos7_ticlang/Stack/Config/znp_cnf.opts...
Checking patch workspace/znp_LP_CC1352P7_4_tirtos7_ticlang/znp.syscfg...
Hunk #3 succeeded at 178 (offset -1 lines).
Checking patch workspace/znp_LP_CC2652R7_tirtos7_ticlang/.project...
Checking patch workspace/znp_LP_CC2652R7_tirtos7_ticlang/Stack/Config/znp_cnf.opts...
Checking patch workspace/znp_LP_CC2652R7_tirtos7_ticlang/znp.syscfg...
Hunk #3 succeeded at 172 (offset -1 lines).
Checking patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/.project...
Checking patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/Stack/Config/znp_cnf.opts...
Checking patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/znp.syscfg...
Hunk #3 succeeded at 173 (offset -1 lines).
Applied patch source/ti/zstack/boards/cc13x2_cc26x2/cc13x2_cc26x2_tirtos7_ticlang.cmd cleanly.
Applied patch source/ti/zstack/boards/cc13x2x7_cc26x2x7/cc13x2x7_cc26x2x7_tirtos7_ticlang.cmd cleanly.
[... 33 lines cut ...]
Applied patch workspace/znp_LP_CC2652R7_tirtos7_ticlang/Stack/Config/znp_cnf.opts cleanly.
Applied patch workspace/znp_LP_CC2652R7_tirtos7_ticlang/znp.syscfg cleanly.
Applied patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/.project cleanly.
Applied patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/Stack/Config/znp_cnf.opts cleanly.
Applied patch workspace/znp_LP_CC2652RB_tirtos7_ticlang/znp.syscfg cleanly.
```

... and the rejected hunk is content of this file : **C:\ti\simplelink_cc13xx_cc26xx_sdk_7_40_00_77\workspace\znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang\znp.syscfg.rej**

```
diff a/workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg b/workspace/znp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/znp.syscfg	(rejected hunks)
@@ -4,6 +4,10 @@
  * @cliArgs --board "/ti/boards/CC1352P_2_LAUNCHXL" --rtos "tirtos7" --product "simplelink_cc13xx_cc26xx_sdk@7.10.00.98"
  * @versions {"tool":"1.15.0+2826"}
  */
+scripting.excludeFromBuild("ti_devices_config.c");
+scripting.excludeFromBuild("ti_radio_config.c");
+scripting.excludeFromBuild("ti_radio_config.h");
+scripting.excludeFromBuild("ti_drivers_config.h");
 
 /**
  * Import the modules used in this configuration.
```

Any help on this topic ? 🙄

amidarius, 2024-03-20:
Just remember, never ever install SDK and IDE when you're tired ! 😄
Wrong simplelink and CCS versions. Installed the latest one, somehow ovelooked that. 🙄
Installed simplelink_cc13xx_cc26xx_sdk_7_10_00_98 + Code Composer Studio 12.3.0.00005 and everything went fine, as expected.
Few hours lost but lession (And lot of other things.) learned. 👌¯\_(ツ)_/¯

closed 2024-04-27 by github-actions[bot]