Koenkk/Z-Stack-firmware #495: Unknown Record Type 3 Error

AngeloAna, 2024-03-11:
I am trying to flash a Sonoff Zigbee Dongle P that I just purchased using the TI SmartRF Flash Programmer 2. It keeps failing with the following errors:

>Initiate access to target: COM3 using 2-pin cJTAG.
>Reading file: C:/Users/Angelo/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20230507/CC1352P2_CC2652P_launchpad_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.

AngeloAna, 2024-03-11:
Picture of settings:

[image]

AngeloAna, 2024-03-12:
A quick update for those having the same issue. I used the python method to update the firmware and was able to get it to work. It was significantly harder to do it this way but I am glad it worked.

closed 2024-03-12 by AngeloAna

kevincaradant, 2024-07-20:
Hi

In fact it's because this software supoport only .bin format and not .hex. 
Just found a converter on internet (https://hexed.it/) and it will work. 

I got this issue today and fix it like that ;)

TimoHess, 2024-12-19:
> Hi
> 
> In fact it's because this software supoport only .bin format and not .hex. Just found a converter on internet (https://hexed.it/) and it will work. I got this issue today and fix it like that ;)
> 
> To convert it from the link
> 
> Click on "ouvrir" and load your .hex file After, you just need to click on "enregistrer sous" and it will be export as .bin file

Shuuush. This was the solution after searching 3 hours what the problem was...... Because the SmartRF Flash Programmer 2 was able to connect to the device.

After converting .hex to .bin it worked under windows like a charm. Thank you very much for your hint.