Koenkk/Z-Stack-firmware #501: Create a firmware with LED support for the latest 20230507.

adamnet01, 2024-05-06:
Could someone create a firmware with LED support for the latest 20230507, as in this version from the link?
https://github.com/egony/cc2652p_cc1352p_RF-STAR/tree/main/firmware/coordinator
The version from the link was originally in a device with 3 LEDs.

_Originally posted by @adamnet01 in https://github.com/Koenkk/Z-Stack-firmware/issues/422#issuecomment-2096834240_

egony, 2024-05-06:
For RF-STAR modules (and for other launchpad based) use JetHome Firmware with right LED support.

adamnet01, 2024-05-06:
is it enough to upload the latest version? What is this patch for?

adamnet01, 2024-05-13:
> W przypadku modułów RF-STAR (i innych opartych na starterze) użyj oprogramowania JetHome Firmware z odpowiednią obsługą diod LED. https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/coordinator/cc2652

is it enough to upload the latest version: CC1352P2_CC2652P_launchpad_coordinator_20230707?
What is this patch for?

adamnet01, 2024-05-13:
> For RF-STAR modules (and for other launchpad based) use JetHome Firmware with right LED support. https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/coordinator/cc2652

is it enough to upload the latest version: CC1352P2_CC2652P_launchpad_coordinator_20230707?
What is this patch for?

closed 2024-06-22 by github-actions[bot]