Koenkk/Z-Stack-firmware #502: TI SmartRF Flash Programmer v2 will not flash file CC2652R_coordinator_20230507.hex

mcmellow, 2024-05-07:
have an Electrolama zig-a-zig-ah! using TI SmartRF Flash Programmer v2 to flash this dongle.  Flashing CC2652R_coordinator_20211217.hex works OK.  However flashing CC2652R_coordinator_20230507.hex ended up in a fail 

```
>Initiate access to target: COM4 using 2-pin cJTAG.
>Reading file: C:/_userdata/Raspberry/ZigBee/CC2652R_coordinator_20211217.hex.
>Start flash erase ...
>Erase finished successfully.
>Start flash programming ...
>Programming finished successfully.
>Start flash verify ...
>Skip verification of unassigned page: 21.
>Skip verification of unassigned page: 22.
>Skip verification of unassigned page: 23.
>Skip verification of unassigned page: 24.
>Skip verification of unassigned page: 25.
>Skip verification of unassigned page: 26.
>Skip verification of unassigned page: 27.
>Skip verification of unassigned page: 28.
>Skip verification of unassigned page: 29.
>Skip verification of unassigned page: 30.
>Skip verification of unassigned page: 31.
>Skip verification of unassigned page: 32.
>Skip verification of unassigned page: 33.
>Skip verification of unassigned page: 34.
>Skip verification of unassigned page: 35.
>Skip verification of unassigned page: 36.
>Skip verification of unassigned page: 37.
>Skip verification of unassigned page: 38.
>Skip verification of unassigned page: 39.
>Skip verification of unassigned page: 40.
>Skip verification of unassigned page: 41.
>Skip verification of unassigned page: 42.
>Page: 0 verified OK.
>Page: 1 verified OK.
>Page: 2 verified OK.
>Page: 3 verified OK.
>Page: 4 verified OK.
>Page: 5 verified OK.
>Page: 6 verified OK.
>Page: 7 verified OK.
>Page: 8 verified OK.
>Page: 9 verified OK.
>Page: 10 verified OK.
>Page: 11 verified OK.
>Page: 12 verified OK.
>Page: 13 verified OK.
>Page: 14 verified OK.
>Page: 15 verified OK.
>Page: 16 verified OK.
>Page: 17 verified OK.
>Page: 18 verified OK.
>Page: 19 verified OK.
>Page: 20 verified OK.
>Page: 43 verified OK.
>Verification finished successfully.
>Reset target ...
>Reset of target successful.

```
```
>Initiate access to target: COM4 using 2-pin cJTAG.
>Reading file: C:/_userdata/Raspberry/ZigBee/CC2652R_coordinator_20230507.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful. 
```
What can I do to update my zzh!

Westcott1, 2024-05-09:
Exactly the same here!

ArtJames, 2024-05-14:
[Solution](https://github.com/Koenkk/Z-Stack-firmware/issues/468#issuecomment-1673523085)

closed 2024-06-22 by github-actions[bot]