Koenkk/Z-Stack-firmware #512: CC1352P2_CC2652P_launchpad_coordinator_20240710.zip has __MACOSX folder in it

esand, 2024-08-25:
The latest release's (and master) zip file has a __MACOSX folder (with a file in it) that causes an error when trying to use the following (common flashing) docker command:

`docker run --rm --device /dev/ttyUSB0:/dev/ttyUSB0 -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip ckware/ti-cc-tool -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb`

The error mentions the unexpected __MACOSX folder and file, as the flashing tool is expecting a single file within the zip.

Previous releases of the zip did not contain any other files. I'm assuming that the inclusion of the __MACOSX folder was unintentional.

jjpvdpoel, 2024-08-25:
It's a Mac OS thing, you can delete it safely

esand, 2024-08-25:
> It's a Mac OS thing, you can delete it safely

Yes, however the script that downloads the file (in the container in the example command line I pasted) is not designed to delete/ignore the folder, and it causes an error. The container I referenced is fairly popular and the command is likely used by lots of people. This limits people to downloading and extracting the .hex file from the zip which is less convenient than being able to specify a URL directly to the zip.

leotantantan, 2024-08-26:
I've got the same issue.

HaareistNase, 2024-08-26:
Im using Zigstar-Addon in Home-Assist to update firmware, and this Addon will fail too with the maxos-drawer in Zip.
I download the file, delete the macos-drawer and upload the file to my personal webspace and use this url, than the addon work.

Thank you for this thread. Without I had no clue how to solve zip "short read", because I know, that the zip is not "faulty".

Koenkk, 2024-08-26:
Fixed!

closed 2024-08-26 by Koenkk