Koenkk/Z-Stack-firmware #513: Error when flashing Z-Stack_3.x.0_coordinator_20240710

adamlc, 2024-08-27:
I have just flashed Z-Stack_3.x.0_coordinator_20240710 to my Sonoff Dongle. I got the following error, but saying that it appears to have upgraded and everything is still working!

```
docker run --rm \
>     --device /dev/ttyUSB0:/dev/ttyUSB0 \
>     -e FIRMWARE_URL=https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip \
>     ckware/ti-cc-tool -ewv -p /dev/ttyUSB0 --bootloader-sonoff-usb
Downloading firmware from https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_launchpad_coordinator_20240710.zip
Firmware source: 'CC1352P2_CC2652P_launchpad_coordinator_20240710.zip'
Firmware file:   'CC1352P2_CC2652P_launchpad_coordinator_20240710.hex'
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:25:7C:AE:5E
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
Write 104 bytes at 0x00057F988
    Write done
Verifying by comparing CRC32 calculations.
    Verified (match: 0xd9dd0124)
ERROR: int() can't convert non-string with explicit base
```

Firmware shown in Z2M:

[image]

Koenkk, 2024-08-27:
> ERROR: int() can't convert non-string with explicit base

I also got this, looks like a bug in CC2538-bsl, I suggest opening an issue there (firmware is flashed correctly)

closed 2024-08-27 by Koenkk

agross, 2024-08-31:
@adamlc https://github.com/git-developer/ti-cc-tool/issues/13#issuecomment-2239050506