Koenkk/Z-Stack-firmware #531: Network crash with BUFFER_FULL errors after a few hours (fw 20240710)

raulbalanza, 2024-12-17:
I have a Zigbee network with around 55 devices. At first I flashed my adapter (ZBDongle-P) with the 20230507 firmware, and I located the coordinator in a corner of the house. When I did that and after pairing all the devices, the network became unstable after some hours. Restarting Z2M solved the issue, but it was very unstable.

Then, when the 20240710 firmware got released, I flashed my adapter with that version, hoping that the instability would be fixed. However, what happens now is that the network crashes after a few hours with several **0x11: BUFFER_FULL** errors. However, now restarting Z2M does not fix the issue and I have to unplug and replug the adapter (apart from restarting Z2M).

I thought that maybe the location of the adapter was the problem, as maybe it had some signal strength problems. I then moved it to a centered place in the house, but the problem persists. Now I do not know what to do.

My network is composed of the following devices: TS0001: 52, TS130F: 5, SNZB-02D: 3, TS0601: 2 (most of them are routers).
I attach the network logs of the last Z2M execution, where the moment when it starts throwing errors can be seen.

Any help would be appreciated :)

[z2m_logs.zip](https://github.com/user-attachments/files/18170883/z2m_logs.zip)

Koenkk, 2024-12-20:
Could you try with this fw: https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226 ?

raulbalanza, 2024-12-23:
Flashed it today, i'll report how it goes :)

raulbalanza, 2024-12-29:
For now it works perfectly, thank you for your help 👌

closed 2024-12-29 by raulbalanza

dinde, 2025-02-03:
Hello,

Exactly the same issue. I downgraded as advised but I'm facing many lag & disconnections.
My automations (z2m) are now working erraticly or lag a lot.
BUFFER_FULL was the initial issue.

Any advise ?
Opening a new issue.
Thanks

fitamix, 2025-02-03:
> Hello,
> 
> Exactly the same issue. I downgraded as advised but I'm facing many lag & disconnections. My automations (z2m) are now working erraticly or lag a lot. BUFFER_FULL was the initial issue.
> 
> Any advise ? Opening a new issue. Thanks

I was using zigbee2mqtt and events were getting lost or going very slow. I have changed my integration to zha and now events are coming much better and faster.

dinde, 2025-02-05:
> I was using zigbee2mqtt and events were getting lost or going very slow. I have changed my integration to zha and now events are coming much better and faster.

Thanks for your answer.
zigbee2mqtt is not the problem. It's a coordinator firmware issue here.
I don't use ZHA at all and do not plan using it.
My automations are within zigbee2mqtt not on HA.
Latest firmware was working perfectly until it bugs.
2022 firmware is giving lag and events lost.

rmi1974, 2025-02-10:
Same here. Have Dongle-P and network with 120 zigbee devices.

For the first time I lost half of my network today with the mentioned error.
Never had this before, with old firmware (2022).
I had the firmware 20240710 flashed since October 2024.

Very weird. Flashed now the old 2022 firmware.

Gecko3600, 2025-02-11:
same happend to me:
https://github.com/Koenkk/zigbee2mqtt/issues/25945