Koenkk/Z-Stack-firmware #534: Instructions for flashing a coordinator

Boerny41, 2025-01-27:
Hi,

I got a cc2652p, but the default firmware by Alex Egony is no longer maintained. I wanted to flash this firmware, but I struggle to find instructions.
The web page of the coordinator (HamGeek Zigbee 3.0) states that either yours or Egonys hardware can be used, so it should be compatible.

"You can use the hardware by Koenkk (Koen Kanters) https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator or by Egony (Alex Egony) [https://github.com/egony/cc2652p_E72 -2G4M20S1E/tree/master/firmware/coordinator](https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator). Use Egony's coordinator firmware by default."

popy2k14, 2025-01-27:
hi, i do it this way directly on my HA machine:
 * connect with SSH
 * download https://github.com/JelmerT/cc2538-bsl
 * cd ~/share/cc2538-bsl
 * Download latest file: https://github.com/Koenkk/Z-Stack-firmware/releases
   wget ....
   !! use the correct one for your stick !!
 * unzip CC1352P2_CC2652P_launchpad_coordinator_20230922.zip
 * rm CC1352P2_CC2652P_launchpad_coordinator_20230922.zip
 * stop Z2M in HA
 * python3 cc2538-bsl.py -e -v -w --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
 * start Z2M in HA -> in Z2M "About" you should see the new firmware

Boerny41, 2025-01-27:
Uff that's more complicated than expected. My `~/share/` folder seems to be empty.

popy2k14, 2025-01-28:
You can use this addon from zigstar, see here: https://www.youtube.com/watch?v=E39ECyCXJaA

Should work with most sticks and is easier 😃
Found yesterday but have not tried it myself yet.

Koenkk, 2025-01-28:
You can flash it from your web browser nowadays: https://smlight.tech/flasher/#other_cc

closed 2025-03-09 by github-actions[bot]