Koenkk/Z-Stack-firmware #535: Sonoff Dongle P Firmware 20240710 BUFFER_FULL and all devices disconnected

dinde, 2025-02-03:
Hello,

Exactly the same issue as #531 
I downgraded as advised to 20221226 but I'm facing many lag & disconnections now.
My automations (z2m) are now working erraticly or lag a lot.

BUFFER_FULL is the initial issue on firmware 20240710

Any advise or am I doing something wrong ?
Thanks

Gecko3600, 2025-02-11:
i can confirm this exact situation: https://github.com/Koenkk/zigbee2mqtt/issues/25945

Koenkk, 2025-02-12:
Could you try with https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226 ?

dinde, 2025-02-18:
Hello,
Thanks for your answer @Koenkk .
As stated on my original post, I tried to downgrade to this version as advised on #531 
I get problems as well, strong lags, missed events, unusable even with automations within z2m.
For now I'm on https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20230507

thogens, 2025-02-26:
This is so strange and I faced similar situations with my Dongle P. Now on another hardware SLZB-06, but guess what, after updating to 20240710 problems started again with the new hardware. (Before upgrading it worked PERFECTLY and now there are all the issues again... And downgrading to 2022... is not really doing the trick.
There must be something that even after downgrading persists in some way.
How about 2023 FW? Does it work longer than a day?

SolusCado, 2025-03-26:
I've started having this problem as well. I replaced my hardware, thinking that was the problem. but after updating the stock firmware 
to yours (20240710) and connecting it to HA, virtually all of my devices are listed as "unavailable", and I'm seeing buffer_full messages in the logs.

Koenkk, 2025-03-26:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/discussions/545

closed 2025-03-26 by Koenkk

SolusCado, 2025-03-27:
For my part, the issue ended up being a corrupted zigbee database. Recreating it has solved my issues.