Koenkk/Z-Stack-firmware #538: [REQUEST] Build and release updated Z-Stack_3.x.0 Zigbee Router firmware for CC2652, CC1352, and the new CC2674 and CC1354 chips?

Hedda, 2025-02-12:
Please consider building/releasing updated Z-Stack_3.x.0 **Zigbee Router** firmware based on the very latest SimpleLink SDK from Texas Instruments:

* https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin
  * https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0

Current **Zigbee Router** builds are compatible with TI's CC26x2/CC13x2 but not with new CC26x4 (CC2674P/CC2674R) and CC13x4 (CC1354P) chips:

* https://github.com/Koenkk/Z-Stack-firmware/discussions/476

Not checked the changelog history for CC26x2/CC13x2 but hopefully there might even be some zstack bug-fixes for those too related to routing(?).
.
Anyway, hardware example is the CC2674P (CC2674P10) based "SLZB-07p10" USB dongle and "SLZB-06p10" Ethernet/WiFi adapters from SMLIGHT:

* SLZB-07p10 -> https://smlight.tech/product/slzb-07p10/

* SLZB-06p10 -> https://smlight.tech/product/slzb-06p10/

PS: SMLIGHT is shipping hardware already but there are no Zigbee firmware images shared in their GitHub repos yet:

* https://github.com/smlight-dev

PPS: Off-topic however for reference; note that Koenkk do build OpenThread firmware images for all CC26XX (including CC2674P and CC2674R):

* https://github.com/Koenkk/ot-ti/t

Hedda, 2025-03-15:
Still wanted

backlink-market, 2025-03-25:
please compile a router version, thanks

Koenkk, 2025-03-25:
Will do it once I have some confidence in the new SDK: https://github.com/Koenkk/Z-Stack-firmware/discussions/545

Koenkk, 2025-04-09:
Available now! https://github.com/Koenkk/Z-Stack-firmware/discussions/546

closed 2025-04-09 by Koenkk

Hedda, 2025-04-09:
Half closed since no CC2674?

Koenkk, 2025-04-09:
Still wip, see https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-12782959

Onepamopa, 2025-04-18:
Looking for newer ROUTER version for CC2652P (SLZB-06).

Current one is 3 years old now... no updates !?

Hedda, 2025-04-18:
@Onepamopa see -> https://github.com/Koenkk/Z-Stack-firmware/discussions/546