Koenkk/Z-Stack-firmware #540: SLZB-06 - 20240710 - problem with BUFFER_FULL - downgrade required to solve

giopeco, 2025-02-15:
Hi,
I've a **SLZB-06** adapter running with HA zigbee2mqtt.

**Adapater**
Core 2.7.1
FW coordinator: 20240710

**Home Assistant**
Core 2025.2.4
Supervisor 2025.02.1
Frontend 20250214.0

**Zigbee2mqtt**: 2.1.1-1

This morning at 5:00am the network collapsed. Log of z2m shows error: **failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'**

I tried to restore my HA backup of yesterday: no result.

I tried to downgrade adapter core with no results.

I downgraded adapter firmware to 20221226, **problem solved**, network back online.

Working conf:

**Adapater
Core 2.7.1
FW coordinator: 20240710**

giopeco, 2025-02-17:
I just updated, I'll let you know.

20221226-->20230507-->20230922-->20240315-->20240315-->20240710

giopeco, 2025-02-18:
Hi, the  issue happened after few minutes 

atached the log of last 1000 lines

[45df7312_zigbee2mqtt_2025-02-18T06-27-51.473Z.log](https://github.com/user-attachments/files/18839985/45df7312_zigbee2mqtt_2025-02-18T06-27-51.473Z.log)

giopeco, 2025-02-18:
i will try again this evening, to make it work you also have to uninstall and reinstall z2mqtt.

giopeco, 2025-02-18:
ok, i've just:

- stopped z2m
- updated SLZB-06 TO 20240710
- Started z2m

started working properly at 19:50, I will keep you updated.

thogens, 2025-02-26:
Is there any update? 
Interestingly I moved from Sonoff P to SLZB-06 and with stock firmware of 20221226 the network just worked perfectly and rock solid for 3 weeks!
Accidentally I flashed 20221226 over the old one and got some error.
I was curious and updated to 20240710 and I was getting the exact same issues, I had with Sonoff P as well. -> Routers leaving network and overall instability.
Rolling back to 20221226 solves the issue for now and I'm looking closely to any misbehaviour. (as with the Dongle-P I also thought rolling back solves everything, but the instabilities came back)

giopeco, 2025-02-26:
hello, for me at the moment the system is stable

ibtangow, 2025-03-01:
Same problem with **CC1352P2_CC2652P_launchpad_coordinator_20240710**. Buffer full and of the entire zigbee network.
I had to downgrade to the version **CC1352P2_CC2652P_launchpad_coordinator_20230507**.
I hope this version keeps the system stable.

Koenkk, 2025-03-01:
Could you try with this fw: 

[CC1352P2_CC2652P_launchpad_coordinator_99250223.hex.zip](https://github.com/user-attachments/files/19040724/CC1352P2_CC2652P_launchpad_coordinator_99250223.hex.zip)

ibtangow, 2025-03-02:
I updated the firmware to this version CC1352P2_CC2652P_launchpad_coordinator_99250223.hex and it reports this error: f**ailed: 'Error: ZCL command**

error 2025-03-02 08:36:22z2m: Publish 'set' 'state' to '****'  0x38398ffffedfd8d8/1 closuresWindowCovering.stop({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:36:28z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c1384e301c99e4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:36:34z2m: Publish 'set' 'brightness' to '****' failed: 'Error: ZCL command 0xa4c1384c8185dbd1/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:36:40z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c13803aa0615ee/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:36:46z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c138015fe049ab/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:36:58z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c1381b3f3136ea/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:04z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c13882b5df8b04/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:10z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c1384e301c99e4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:16z2m: Publish 'set' 'brightness' to '****' failed: 'Error: ZCL command 0xa4c1384c8185dbd1/11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:22z2m: Publish 'set' 'state' to '****a' failed: 'Error: ZCL command 0xa4c13803aa0615ee/11 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:28z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c138015fe049ab/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:40z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c1381b3f3136ea/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:46z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c13882b5df8b04/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:52z2m: Publish 'set' 'state' to '****' failed: 'Error: ZCL command 0xa4c1384e301c99e4/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
error 2025-03-02 08:37:58z2m: Publish 'set' 'brightness' to '****' failed: 'Error: ZCL command 0xa4c1384c8185dbd1/11

ibtangow, 2025-03-02:
I have updated to version CC1352P2_CC2652P_launchpad_coordinator_99250223.hex, after several errors at startup (attached files).

[z2m_errors.txt](https://github.com/user-attachments/files/19044158/z2m_errors.txt)

I had to reinstall zigbee2mqtt and then it booted fine. 
I'll let it run for a few days to see the result.

ibtangow, 2025-03-04:
After 2 days working fine, the network went down tonight. 
Checking several devices failed with status **'(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'**

[log_z2m_20250304.txt](https://github.com/user-attachments/files/19075387/log_z2m_20250304.txt)

Koenkk, 2025-03-04:
Could you try with `99250304` from https://github.com/Koenkk/Z-Stack-firmware/issues/518#issuecomment-2698726919?

giopeco, 2025-03-04:
I updated to Zigbee2MQTT 2.1.3-1 yesterday
No problem at the moment.

Adapater
Core 2.7.1
FW coordinator: 20240710

ibtangow, 2025-03-04:
I have updated to **99250304**

(z2m version: 2.1.3-1)

I'll let it run for a few days to see the result.

ibtangow, 2025-03-09:
At the moment, working fine.

qraynaud, 2025-03-09:
I had the same issue with **20240710**, just flashed **99250304** and it seems to work like a charm! Thanks @Koenkk!

qraynaud, 2025-03-09:
Correction: after some time running it, it's still problematic. I have no more `BUFFER_FULL` errors but the network is very very slow. When I click in the UI to toggle a light, it can take up to 30 seconds to achieve it. Any idea?

qraynaud, 2025-03-10:
I let it run hoping it would stabilize but no. It's even worse. Now I just can't even get a single device to change state using the Z2M UI. I’ll try to revert the fw to an older version for now because I'm stuck with lights that are turned on 🤣.

Allenspeed1, 2025-03-12:
Mine is on docker, but single virtualization.

ibtangow, 2025-03-13:
@qraynaud try to delete file backup-coordinator.json:

1. Disable Z2M

2. Delete backup-coordinator.json.

3. Restart Z2M.

Koenkk, 2025-03-13:
@ibtangow it turns out the SLZB06 requires the "other" firmware instead of the "launchpad" one. Could you try the 20240710 other firmware? https://github.com/Koenkk/Z-Stack-firmware/releases/download/Z-Stack_3.x.0_coordinator_20240710/CC1352P2_CC2652P_other_coordinator_20240710.zip

ibtangow, 2025-03-13:
I have TubesZB CC2652 Ethernet. And for the moment, 99250304 working fine.

thogens, 2025-03-13:
@Koenkk: isn’t the web interface of ZLZB06 fetching the correct firmware?

Koenkk, 2025-03-15:
@thogens I confirmed that it is

drbartek, 2025-03-17:
> [@thogens](https://github.com/thogens) I confirmed that it is

@Koenkk  Is there a version “99250304” for the “other_coordinator” firmware (SLZB-06) to download?

Koenkk, 2025-03-17:
@drbartek 

[CC1352P2_CC2652P_other_coordinator_99250304.hex.zip](https://github.com/user-attachments/files/19297541/CC1352P2_CC2652P_other_coordinator_99250304.hex.zip)

drbartek, 2025-03-17:
> [@drbartek](https://github.com/drbartek)
> 
> [CC1352P2_CC2652P_other_coordinator_99250304.hex.zip](https://github.com/user-attachments/files/19297541/CC1352P2_CC2652P_other_coordinator_99250304.hex.zip)

Thanks!

ibtangow, 2025-03-24:
with firmware **fw: 20221226**, everything is perfect from the start. 
The devices remain stable on the network!