Koenkk/Z-Stack-firmware #564: Error when trying to flash CC1352P2_CC2652P_launchpad_coordinator_20250321.zip on Sonoff dongle P

caseca, 2025-07-10:
Hello, I use HassOS and flash my Sonoff dongle P with the ZigStar TI CC2652P/P7 FW Flasher add-on. When I tried to flash the latest version CC1352P2_CC2652P_launchpad_coordinator_20250321.zip, I got the following error: "ERROR: 'utf-8' codec can't decode byte 0xa3 in position 939: invalid start byte". 

Here's the complete log:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting
-----------------------------------------------------------
 Add-on: ZigStar TI CC2652P/P7 FW Flasher
 ZigStar TI CC2652P/P7 firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.4.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 16.0  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2025.7.1
 Home Assistant Supervisor: 2025.07.1
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service cc2652-flasher: starting
[11:25:59] INFO: Starting CC2652P flasher with Sonoff /dev/ttyUSB0
Setting filename to /root/firmware.hex
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from /root/firmware.hex
Your firmware looks like an Intel Hex file
ERROR: 'utf-8' codec can't decode byte 0xa3 in position 939: invalid start byte
[11:26:00] INFO: cc2652-flasher-up script exited with code 1
s6-rc: warning: unable to start service cc2652-flasher: command exited 1
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Thanks in advance for any help.

gvillo, 2025-07-11:
same using custom url: [https://github.com/Koenkk/Z-Stack-firmware/raw/refs/tags/Z-Stack_3.x.0_coordinator_20250321/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20250321.zip](https://github.com/Koenkk/Z-Stack-firmware/raw/refs/tags/Z-Stack_3.x.0_coordinator_20250321/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20250321.zip)

gvillo, 2025-07-11:
I was able to flash it using https://dongle.sonoff.tech/sonoff-dongle-flasher/, so the link that @Koenkk posted should work too as well. The pitfall was disconnecting the dongle and connect it to the PC.

I do understand too that the add-on is not being updated at all, but it was working using raw URLs until 2024 firmware.

caseca, 2025-07-11:
Hey, I used the add-on specifically to avoid removing the dongle from my server, and it's a shame I can't do that anymore. I'll try your recommended tools as soon as I can and let you know. Thanks a lot.