Koenkk/Z-Stack-firmware #571: Consistent 'Buffer Full' Error in Zigbee2MQTT – Anyone Else Experiencing This?

teotho, 2025-07-29:
Hi,

I'm facing the BUFFER_FULL problem with 20250321 version. ( Image Below )

[image]

### Current Configuration 
Home Assistant with Zigbee2MQTT addon
**Home Assistant Current version ** : 2025.4.4
**Zigbee2MQTT addon version** :  2.4.0-1
I'm using SLZB-06 with Zigbee Core OS version : 2.8.7, Zigbee SoC cc2652P, adapter: Zstack

[image]
[image]

I tries to update to 20250325 but having the same BUFFER_FULL errors

I'm including the full logs if it may help:
[45df7312_zigbee2mqtt_2025-06-06T00-54-06.268Z.log](https://github.com/user-attachments/files/20620122/45df7312_zigbee2mqtt_2025-06-06T00-54-06.268Z.log)

_Originally posted by @teotho in https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-13385091_

Koenkk, 2025-07-29:
This logging only contains the error log, could you provide the debug log from starting z2m until the error? Also make sure to disable the tuya default response (as explained in https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-13881351)

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

teotho, 2025-07-31:
> This logging only contains the error log, could you provide the debug log from starting z2m until the error? Also make sure to disable the tuya default response (as explained in [#545 (reply in thread)](https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-13881351))
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Hello @Koenkk , thank you for your reply. I did enabled the "debug" mode. Here is the logs files. Thank you for your time!

[45df7312_zigbee2mqtt_2025-07-31T13-41-10.708Z.log](https://github.com/user-attachments/files/21531638/45df7312_zigbee2mqtt_2025-07-31T13-41-10.708Z.log)

teotho, 2025-07-31:
hello @Koenkk , 

As an addition to what have been previously discussed in https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-13866980 . I've seen that @matlab22, was able to set `DISABLE_TUYA_DEFAULT_RESPONSE=true`  as he is using z2m as a systemd service. Which is not my case ( I'm using z2m addon on a HomeAssistant instance installed on a raspberry device)

How can I set `DISABLE_TUYA_DEFAULT_RESPONSE=true` without switching to the dev branch (edge version) in my case?

Thanks for your reply

Reference of https://github.com/Koenkk/Z-Stack-firmware/discussions/545#discussioncomment-13866980 in #545  @mrmaximas

Koenkk, 2025-08-01:
- Did `20250325 ` work? 
- Can you answer this: 
> it seems that the BUFFER_FULL immediately occurs when the first command is send? (so not after a few hours of uptime?)

teotho, 2025-08-02:
Thank you for you reply,

- The `20250325` didn't work ( I got the BUFFER_FULL error immediately after restarting the z2m HA addon )
- I downgraded to  `20221226` and for now there is no BUFFER_FULL error ( I will see if there is any after some hours of run time )

If I don't have any BUFFER_FULL with this configuration I will give you an update here.

One more thing @Koenkk please, can you tell me how to do this   ( FYI : I'm using z2m  v2.5.1-1 as addon on Home Assistant ) :
> How can I set `DISABLE_TUYA_DEFAULT_RESPONSE=true` without switching to the dev branch (edge version) in my case?

Thank you for your time

Koenkk, 2025-08-12:
If you enable `Show unused optional configuration options` you will get `disable_tuya_default_response`

[image]

teotho, 2025-08-13:
Hello @Koenkk,

Thank you for your reply, I was able to set `DISABLE_TUYA_DEFAULT_RESPONSE=true`. Untill now there is no `BUFFER_FULL` error or similar errors to what I had show you in the previous post https://github.com/Koenkk/Z-Stack-firmware/issues/571#issuecomment-3146559431 . I will attach the logs file so you can have a look at it if you want.

I also have some questions about this : 

- How did disabling Tuya Default response help with `BUFFER_FULL` errors?   ( How does this option affect the normal functioning of ZHA z2m) 
- Is this affecting the communication performance of my Tuya devices with ZHA Zigbee2Mqtt?

[45df7312_zigbee2mqtt_edge_2025-08-13T14-15-16.170Z.log](https://github.com/user-attachments/files/21756556/45df7312_zigbee2mqtt_edge_2025-08-13T14-15-16.170Z.log)

I will update you in the next upcoming 24h if I got any errors or not.

Thank you for your time and reply

Koenkk, 2025-08-14:
> How did disabling Tuya Default response help with BUFFER_FULL errors? ( How does this option affect the normal functioning of ZHA z2m)

The default response is a command send to acknowledge it has been received by z2m, but most likely the Tuya devices don't care about it.

> Is this affecting the communication performance of my Tuya devices with ZHA Zigbee2Mqtt?

If you haven't seen any issues by now it should be good.

teotho, 2025-08-15:
hello @Koenkk,

Thanks for the explanation. Very appreciated.

I'm writing this comment to update you about the situation.

I get `failed with error: 'SRSP - ZDO - mgmtPermitJoinReq after 6000ms'` errors when I permit join on the z2m edge instance ( with `DISABLE_TUYA_DEFAULT_RESPONSE` option set to true of course) as you can see in this screenshot : 

[image]

You have said in previous thread that this error is just an other facet of the same problem ( the `BUFFER_FULL `error one )

I will attach the full logs file with debug enabled if you want to look into it : 

[45df7312_zigbee2mqtt_edge_2025-08-15T00-58-54.583Z.log](https://github.com/user-attachments/files/21782223/45df7312_zigbee2mqtt_edge_2025-08-15T00-58-54.583Z.log)

What do you think? Any further suggestions?

Thank you for your time and your reply.

teotho, 2025-08-19:
Hello @Koenkk ,

Thanks you for your time and reply. 

Is the criteria of saying that a device X is spammy based on the number of the occurrences of RX_events (`Received Zigbee message from '<device>'`) in the log files (Over a long period of time of course ) or by accessing the information provided in the z2m HealthCheck on the WindFront?

With regard to the buffer full error, is the contributing factor the NumberOfMessages or the MessagesPerSecond rate of the devices (as indicated in the Zigbee2Mqtt Wind Front)? If so, what would be the healthy or maximum NumberOfMessages or MessagesPerSecond rate for my coordinator (SLZB-06 with a cc2652p Zigbee SoC chip)?

One last thing is that I've connected a second coordinator ( Identical model and configuration
 to the first ) and I'm planning to reduce the devices load of the first coordinator by pairing half of the devices (58 devices) to this second coordinator. What do you think about this approach? is it even relevant or not?

Again, thank you for your time and reply.

Koenkk, 2025-08-19:
The limits likely depend on how to network is formed (e.g. if coordinator has more childs it might crash earlier) so I dont have a definite answers for this. Splitting your network in 2 should help a lot.