Koenkk/Z-Stack-firmware #573: 0x11 BUFFER FULL

Toutzn, 2025-08-09:
After getting the new firmware I got the same issue:

```
BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:262:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1049:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:505:35)
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:439:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:437:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:86:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1000:28)
[2025-08-09 08:38:08] debug: 	zh:controller:endpoint: ZCL command 0x70b3d52b60125676/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-08-09 08:38:08] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x70b3d52b60125676:6715/1 (0,0,1)
[2025-08-09 08:38:08] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":6715,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":39,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,159,0,5,5,8,5,11,5]}}
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,59,26,1,1,4,11,39,0,30,9,16,159,0,5,5,8,5,11,5,164]
[2025-08-09 08:38:08] debug: 	zh:controller:endpoint: ZCL command 0x70b3d52b6010172c/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-08-09 08:38:08] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x70b3d52b6010172c:7213/1 (0,0,2)
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,17,117]
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,17,117]
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [17] - 117
[2025-08-09 08:38:08] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":17}
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 08:38:08] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":7213,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":40,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,160,0,5,5,8,5,11,5]}}
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,45,28,1,1,4,11,40,0,30,9,16,160,0,5,5,8,5,11,5,132]
[2025-08-09 08:38:08] debug: 	zh:controller:endpoint: Error: ZCL command 0x70b3d52b60125676/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":6715,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":39,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,159,0,5,5,8,5,11,5]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:262:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1049:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:505:35)
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:439:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:437:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:86:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1000:28)
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,17,117]
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,17,117]
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [17] - 117
[2025-08-09 08:38:08] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":17}
[2025-08-09 08:38:08] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 08:38:08] debug: 	zh:controller:endpoint: Error: ZCL command 0x70b3d52b6010172c/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":7213,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":40,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,160,0,5,5,8,5,11,5]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:262:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1049:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:505:35)
    at /app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:439:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:437:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:86:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:347:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1000:28)
```

Zigbee2MQTT version
[2.6.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.0)

ZStack3x0
0x00124b0030d05e14
Revision: 20250321

ID_MODEL: Sonoff_Zigbee_3.0_USB_Dongle_Plus
ID_MODEL_ENC: Sonoff\x20Zigbee\x203.0\x20USB\x20Dongle\x20Plus

Toutzn, 2025-08-09:
This is from then container app:

21:47:14] INFO: Preparing to start...
[21:47:14] INFO: Socat not enabled
[21:47:15] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-08-09 21:47:16] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-08-08.16-19-31'
[2025-08-09 21:47:16] info: 	z2m: Logging to console, file (filename: log.log)
[2025-08-09 21:47:16] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-08-09 21:47:16] info: 	z2m: Starting Zigbee2MQTT version 2.6.0 (commit #unknown)
[2025-08-09 21:47:16] info: 	z2m: Starting zigbee-herdsman (5.0.4)
[2025-08-09 21:47:16] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":26755,\"extendedPanID\":[115,217,56,100,206,185,245,105],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a11781cfa1bef1198799ed94909ffd0-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-08-09 21:47:17] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"3a11781cfa1bef1198799ed94909ffd0","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a11781cfa1bef1198799ed94909ffd0-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-08-09 21:47:17] debug: 	zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a11781cfa1bef1198799ed94909ffd0-if00-port0
[2025-08-09 21:47:17] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":26755,"extendedPanID":[115,217,56,100,206,185,245,105],"channelList":[25]},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a11781cfa1bef1198799ed94909ffd0-if00-port0","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-08-09 21:47:17] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3a11781cfa1bef1198799ed94909ffd0-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-08-09 21:47:17] info: 	zh:zstack:znp: Serialport opened
[2025-08-09 21:47:17] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-08-09 21:47:17] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-08-09 21:47:17] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-08-09 21:47:17] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: <-- [254,6,65,128,0,2,1,2,7,1,192]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,6,65,128,0,2,1,2,7,1,192]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: <-- AREQ: SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: <-- [254,2,97,1,89,6,61]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,2,97,1,89,6,61]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: <-- SRSP: SYS - ping - {"capabilities":1625}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: --> SREQ: SYS - version - {}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,2,35]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: <-- [254,10,97,2,2,1,2,7,1,209,254,52,1,0,116]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,10,97,2,2,1,2,7,1,209,254,52,1,0,116]
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,209,254,52,1,0] - 116
[2025-08-09 21:47:18] debug: 	zh:zstack:znp: <-- SRSP: SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20250321}
[2025-08-09 21:47:18] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-09 21:47:18] debug: 	zh:zstack: Adapter concurrent: 16
[... cut ...] 2/Mercator Ikuü,SPP02GIP/Mercator Ikuü,SPP02G/Mercator Ikuü,SPP04G/Mercator Ikuü,MG-GPO01/MakeGood,MG-AUZG01/MakeGood,HG06338/Lidl,HG08673/Lidl,HG06337/Lidl,WP30-EU/LELLKI,WK34-EU/LELLKI,XF-EU-S100-1-M/LELLKI,CM001/LELLKI,WP33-EU/WP34-EU/LELLKI,07752L/Immax,AUT000069/AutomatOn
[2025-08-09 21:47:22] debug: 	zhc: Getting definitions for: ikea.js,91
[2025-08-09 21:47:22] info: 	z2m: zigbee-herdsman started (resumed)

I had a crash right now again. I will try to enable the log of the service itself.

Toutzn, 2025-09-02:
It is very confused. for now it is running well. I had more than 4 crashes at one day before. I will come back to you, if it crashes again.

Koenkk, 2025-09-13:
@Toutzn can you try setting `adapter_concurrent` to `4` and see if that helps? ([docs](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#advanced-configuration))

Toutzn, 2025-09-13:
Ok. Did it. See what happens. Come back to you if I get new informations. Thx

Toutzn, 2025-09-14:
Nope. Same shit again...

[45df7312_zigbee2mqtt_2025-09-14T11-05-02.276Z.log](https://github.com/user-attachments/files/22319518/45df7312_zigbee2mqtt_2025-09-14T11-05-02.276Z.log)

Restart doesn't help. Only unplug and connect USB Stick again helps.

Toutzn, 2025-09-14:
And Stange things are, sometimes it works for a month or more, sometime I need to disconnecting the stick a couple of times per day.

Toutzn, 2025-09-15:
Hi. This morning the same issue again, but no time to restart and capture the logs, because the hole house was very dark.
But I zipped the logs of the last days we had the iss and restarted before unplug an connect the stick again.

Hope, you can find something which helps. Used tar, but upload is not allowed, gziped the tar .

[logs.tar.gz](https://github.com/user-attachments/files/22327245/logs.tar.gz)

MikeVRT, 2025-09-28:
Hello everyone, I have the same problem in the same environment (Proxmox VM for HASS). For years, it worked stably (half-year uptimes), but now it doesn’t for ~ 1 year. The coordinator is based on a CC2652P.
What I could add to the discussion:
1. I thought the issue was caused by the new firmware, but rolling back didn’t help (I tested up to the 20220219 firmware, which was old-style flashed — it used to be 100% stable, but not anymore). So the problem must be due to changes in z2m itself (and how the firmware reacts), I guess.
 
2. Proxmox's Debian generates the following log when a crash occurs:
`Aug 29 21:36:01 virthost QEMU[4482]: kvm: libusb_release_interface: -4 [NO_DEVICE]
Aug 29 21:36:02 virthost kernel: usb 1-8: new full-speed USB device number 5 using xhci_hcd
Aug 29 21:36:02 virthost kernel: usb 1-8: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.32
Aug 29 21:36:02 virthost kernel: usb 1-8: New USB device strings: Mfr=0, Product=2, SerialNumber=0
Aug 29 21:36:02 virthost kernel: usb 1-8: Product: USB Serial
Aug 29 21:36:02 virthost kernel: ch341 1-8:1.0: ch341-uart converter detected
Aug 29 21:36:02 virthost kernel: usb 1-8: ch341-uart converter now attached to ttyUSB0
Aug 29 21:36:03 virthost kernel: ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
Aug 29 21:36:03 virthost kernel: ch341 1-8:1.0: device disconnected
Aug 29 21:36:03 virthost CRON[1652892]: pam_unix(cron:session): session closed for user root
Aug 29 21:36:04 virthost kernel: usb 1-8: reset full-speed USB device number 5 using xhci_hcd
Aug 29 21:36:04 virthost kernel: usb 1-8: USB disconnect, device number 5
Aug 29 21:36:04 virthost kernel: usb 1-8: new full-speed USB device number 6 using xhci_hcd
Aug 29 21:36:05 virthost kernel: usb 1-8: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.32
Aug 29 21:36:05 virthost kernel: usb 1-8: New USB device strings: Mfr=0, Product=2, SerialNumber=0
Aug 29 21:36:05 virthost kernel: usb 1-8: Product: USB Serial
Aug 29 21:36:05 virthost kernel: ch341 1-8:1.0: ch341-uart converter detected
Aug 29 21:36:05 virthost kernel: usb 1-8: ch341-uart converter now attached to ttyUSB0
Aug 29 21:36:06 virthost kernel: ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
Aug 29 21:36:06 virthost kernel: ch341 1-8:1.0: device disconnected
Aug 29 21:36:06 virthost kernel: usb 1-8: USB disconnect, device number 6
Aug 29 21:36:07 virthost kernel: usb 1-8: new full-speed USB device number 7 using xhci_hcd`

3. All crashes are probably caused by messages from the Aqara door sensor (MCCGQ11LM). I'm not completely sure, but they always coincide with open/close messages from it, and the system crashes in the same second.

I understand that, formally, it looks like a hardware crash, but I’ve tested different USB ports and with/without extension cables as well. As @Toutzn said, it might work for a few days or crash twice per day, but the median uptime is only 3–4 days compared to a h/year of stable operation before.

zen2, 2025-10-02:
> and z2m couldn’t start because the VM doesn’t detect “/dev/ttyUSB0,” but the issue isn’t in the VM env. itself, since the host can’t see it either.

As far as i understand, this is the different events in order:
0- Z2M firmware crashes
1- USB device disappears
2- USB device found
3- ch341-uart detected and attached to ttyUSB0
4- ch341-uart disconnect from ttyUSB0
5- reset (only first time)
6- GOTO 2

I'm afraid that the firmware stay a stuck state after the crash and the coordinator need to be unplugged to force a hardware reset (no power clear the coordinator memory).
It means the problem seems correlated to the firmware itself.

As an eventual fix to avoid the physical unplug you can try to force the USB reset from VM or host system:

- using `lsusb` and `usbreset` from `usbutils` package (not available on all distro)
or
- Use `lsub` to find the vendorid/pruductid then
```
device=$(dirname $(grep $VENDORID /sys/bus/usb/devices/*/idVendor | cut -d: -f1))
echo ${device/*\/} > /sys/bus/usb/drivers/usb/unbind
echo ${device/*\/} > /sys/bus/usb/drivers/usb/bind
```

Note that won't cycle the power but unload/load the driver. Some devices stop at driver unloading or reinitialize at driver loading.

There is also another tool "uhubctl" that can simulate a USB disconnect/connect and really power off the usb port if hub is compatible but take care that it will work at USB hub level so that can disconnect/reconnect other USB devices connectes to the same hub.
See: https://github.com/mvp/uhubctl

@Koenkk Is the firmware use a watchdog ?
See: https://dev.ti.com/tirex/explore/content/simplelink_cc13x2_26x2_sdk_4_30_00_54/docs/drivers/doxygen/html/_watchdog_c_c26_x_x_8h.html

Koenkk, 2025-10-02:
@zen2 I don't think the watchdog is enabled but I'm not sure if it something that can just be enabled.

@MikeVRT do you have the debug log when it crashes? (and preferable also up to 1 hour before that)

MikeVRT, 2025-10-02:
> [@MikeVRT](https://github.com/MikeVRT) do you have the debug log when it crashes? (and preferable also up to 1 hour before that)

At z2m log nothing interesting - it just looks like you moved the stick out. Same in reconnection log, if watchdog (z2m) switched on - just "... cannot open /dev/ttyUSB0"

p.s. @zen2 probably meant the watchdog (fw) as the cause of the repeated crashes (if it’s even active).

MikeVRT, 2025-10-02:
> you can try to force the USB reset from VM or host system

I'll check it on the next crash (the usbreset is available in Proxmox by default) and reply here.

zen2, 2025-10-03:
> [@zen2](https://github.com/zen2) I don't think the watchdog is enabled but I'm not sure if it something that can just be enabled.

It seems the watchdog is a TI driver in the SDK level that provides watchdogs functions and so it needs some logic implementation:
https://dev.ti.com/tirex/explore/node?node=A__AKV1fBlIIvnkd6UlidzP2w__com.ti.SIMPLELINK_CC13XX_CC26XX_SDK__BSEc4rl__LATEST

> p.s. [@zen2](https://github.com/zen2) probably meant the watchdog (fw) as the cause of the repeated crashes (if it’s even active).

A watchdog are commonly used for servers at kernel level, it detect if the kernel is stuck (kernel panic, infinite loop, hardware interruption wait) to make the server reboot to avoid to stay stuck. It's based on a simple signal counter that should be reset regularly by the kernel.

So a firmware watchdog should have reset the coordinator like you do by unplug/plug the USB but it seems that your coordinator stay stuck after the crash so it seems either the firmware doesn't use an active watchdog or either it use one that doesn't operate a full firmware reset but only a UART one (USB connection, no power cycle).

Your logs show clearly that the ch341-uart disconnect right after it have reconnect. The ch341-uart is a USB to I2C/SPI/UART converter. I2C/SPI is used to flash an EEPROM and UART to convert serial/parallel signals to USB ones. 

USB devices are often problematic because if the device itself don't manage a reset cycle on driver loading/unloading the system have no way to control the device USB port supply to force it. The system can only load/unload the driver used by the device or power cycle the whole USB hubs if it support correctly the power feature.
That's why most of system issues related to USB devices require a cold boot of the computer.

Ruud-cb, 2025-10-24:
@Koenkk Will do once the problem starts occurring again. NWK_TABLE_FULL is mostly the first issue though.
Wondering though: who throws these errors? Is this core zigbee network errors or z2a exception handling?

Koenkk, 2025-10-24:
`NWK_TABLE_FULL` means that one of the network related tables in the coordinator is full (so it's an error coming from the coordinator, not from one of the other devices in the network)

Ruud-cb, 2025-10-31:
After upgrading and restarting (power off-on), I'm now running stable for 6 days. Don't want to be happy to soon but at least it's not a restart every other day now...
Zigbee2MQTT v2.6.2
Frontend version 2.2.1
zigbee-herdsman-converters version 25.35.1
zigbee-herdsman version 6.1.5
SLZB-06 OS version 3.0.8
SLZB-06 Coordinator firmware version 20250325 (this is NOT from Koenkk, from SLZB itself it looks like).
Home Assistant OS (in Proxmox VM): 2025.10.3

Turned mDNS off within SLZB.

Toutzn, 2026-01-19:
Since I rolled back to 20240710 it seems to be more stable now.

Not sure if some updates of the app fixed it as well. 

Some ideas?

BR. Toto

GaugeTheory, 2026-01-20:
> I remember that I had similar problems with version 20230507, random day/hour the dongle crashed. After update to version 2024 I got a stable working. I use zigbee2mqtt with docker container and I restart the container once per week to refresh everything. After some tests I found that the outage of the dongle was caused by a power failure, I had the dongle connected directly to the rapi4, so I decided to use an usb power hub since then I haven't got failure/outage and the dongle is working 24/7 fine with a lot of traffic/messages per day.

I am having exact same issues. I had upgraded to version 20250321 due to some buffer full errors. But I previously was runnig it not in 20240710. It was even older version. I don't remember upgrading it within last 2 years. Therefore i have no information aobut verison 20240710. 
It was running fine ok for few weeks with 20250321. in the meantime I has increased device count from ~30 to ~80.  After couple of weeks I started to get BUFFER FULL error at a random interval. 

I ran home assistant on rpi4 with z2m 2.7.2. I am clueless to fix the issue. I will try USB3.0 port on rpi4 and maybe try usb power hub. I will update if it helps

popy2k14, 2026-01-21:
@GaugeTheory maybe you have the kernel issue i was facing when i am using rpi4 + sonoff usb dongle plus back than.
If i remember correctly the bug does not come up with an 32 bit os.
My solution was to ditch spi and use an x86 low poer device for HA + Z2M + Dongle.
Here is the reference: https://github.com/Koenkk/zigbee2mqtt/issues/12566#issuecomment-1465295006

I am just assuming the BUFFER_FULL error is an subsequent error, but maybe its not.

As stated above, i am back at 202211226.