Koenkk/Z-Stack-firmware #578: Building firmware : apply patch gives errors.

JulienDevillers, 2025-09-25:
Hello,

I'm trying to build Z-stack coordinator firmware for CC2652RB.
(I think) I followed strictly the compile.md file (Z-Stack_3.x.0):
- installed simple link 8.30.01.01
- installed CCS 20.1.0
- import all requested projects
- updated znp.syscfg files
- If I try to apply the _firmware.patch_ file, I get these errors:
 

```
ju@ju-desktop-ubuntu:~/dev/simplelink_cc13xx_cc26xx_sdk_8_30_01_01$ git apply *.patch --ignore-space-change
firmware.patch:143: trailing whitespace.
#define MAC_CFG_TX_MAX 80 
firmware.patch:163: trailing whitespace.
// Determines the maximum number of “secure partners” that the network Trust Center (ZC) can support. 
firmware.patch:227: trailing whitespace.
    // This does not determine the upper limit in the number of nodes in the network, 
firmware.patch:325: trailing whitespace.
      
firmware.patch:778: trailing whitespace.
  
error: l'application de la rustine a échoué : workspace/znp_LP_EM_CC2674P10_tirtos7_ticlang/.cproject:13
error: workspace/znp_LP_EM_CC2674P10_tirtos7_ticlang/.cproject : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/zr_genericapp.syscfg:5
error: workspace/zr_genericapp_CC1352P_2_LAUNCHXL_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_CC26X2R1_LAUNCHXL_tirtos7_ticlang/zr_genericapp.syscfg:63
error: workspace/zr_genericapp_CC26X2R1_LAUNCHXL_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_CC1352P7_4_tirtos7_ticlang/zr_genericapp.syscfg:65
error: workspace/zr_genericapp_LP_CC1352P7_4_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_CC2652R7_tirtos7_ticlang/zr_genericapp.syscfg:63
error: workspace/zr_genericapp_LP_CC2652R7_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg:65
error: workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_EM_CC2674P10_tirtos7_ticlang/.cproject:13
error: workspace/zr_genericapp_LP_EM_CC2674P10_tirtos7_ticlang/.cproject : la rustine ne s'applique pas
error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_EM_CC2674P10_tirtos7_ticlang/zr_genericapp.syscfg:65
error: workspace/zr_genericapp_LP_EM_CC2674P10_tirtos7_ticlang/zr_genericapp.syscfg : la rustine ne s'applique pas
```
If I look to, for example, to this error:
_error: l'application de la rustine a échoué : workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg:65_
The related diff is:
```
diff --git a/workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg b/workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg
index c28058861..c3422caec 100644
--- a/workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg
+++ b/workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg
@@ -65,6 +65,7 @@ CCFG.enableBootloaderBackdoor = true;
 CCFG.dioBootloaderBackdoor    = 13;
 CCFG.levelBootloaderBackdoor  = "Active low";
 CCFG.srcClkLF                 = "Derived from HF XOSC";
+CCFG.enableDCDC               = false;
 CCFG.ccfgTemplate.$name       = "ti_devices_CCFG_CCFGCC26XXTemplate0";
 
 Display1.$hardware  = system.deviceData.board.components.XDS110UART;
```

Obviously, the code in the file _workspace/zr_genericapp_LP_CC2652RB_tirtos7_ticlang/zr_genericapp.syscfg_ around line 65 does not look like the diff file (I used a screenshot to show line numbers and filename for validation:

[image]

I tried with linux and windows (mainly for errors in .cproject files that are modified by the IDE), but the result is nearly the same (there are some more differences with windows due to access rights).

I should have done something wrong but what?

Thanks
Julien

Koenkk, 2025-09-26:
Did you re-save the syscfg files as described in the readme?

JulienDevillers, 2025-09-29:
Yes but how does this impact the .c source files that do not seem to be consistent with the diff file?

Koenkk, 2025-09-29:
Which `.c` files are not consistent? In the OP I only see `syscfg` files not being consistent.

JulienDevillers, 2025-09-30:
Hum, sorry, I was outside when answering, it's not a matter of .c files but .cproject and zr_genericapp.syscfg.

- .cproject files that fail when applying patches are parts of znp projects. I checked that saving znp.syscfg does not change .cproject file.
- zr_genericapp.syscfg are not intended to be re-saved based on the compile.md file.

That means that all these files are mint from imported example and fails while applying the patch.

Koenkk, 2025-09-30:
> .cproject files that fail when applying patches are parts of znp projects. I checked that saving znp.syscfg does not change .cproject file.

I see this only happens to `zr_genericapp_LP_EM_CC2674P10_tirtos7_ticlang`, I would ignore it for now as this firmware is not officially supported for now (to be fixed later)

> zr_genericapp.syscfg are not intended to be re-saved based on the compile.md file.

That's a mistake indeed, updated the `COMPILE.md`